ruscito/pycomm #4: Connection Recovery

cldodd, 2015-07-31:
Hi ruscito
Sorry, I am a Git novice and newbie to Python but old die hard AB man. Figured (Issues is how to post questions?) I have scripts running between CLX & ML1100 inserting data (Floats/Reals) into mySql on a Raspberry 2. I am impressed and thankful of your efforts in concluding this project - well done. My task is to provide a 24x7 (60 second update/storage) of PLC data for compliance on Water Treatment stations in remote locations. For this reason I have been testing to ensure NOTHING will break the script (my logger script) resulting in loss of data. During the testing I have been making/breaking ethernet connections while polling data and have had several issues where comms is lost and non recoverable. Have you any suggestions on how best to handle make/break situations allowing the CIP connection reestablish gracefully without crashing. Obviously I have rapped some try/except error handling clauses in the script but am struggling with maintaining continuity during abnormal situations. Any suggestions gratefully appreciated Regards Chris

lynnlinse, 2015-07-31:
Likely the socket is 'recoverable', but unless you have made custom TCP keepalive changes in the Linux OS, then it could take HOURS for the lost socket to return your expected errors.

So one thing, look into your global TCP keepalive settings, which generally (always?) host-wide, meaning if you change, it changes for your entire Raspberry. Some systems default to start probes in 2 hours, so no error for 2+ hours!

If you really need sub-1-minute loss detection, you might need to add an external thread/task doing a PING of the PLC once every 30 seconds? The PING will give faster path-loss detection, then you'd need that thread to send a signal to your logger, having it abort the old socket, and start trying to open a new EIP connection. Obviously, if the path is lost (a disconnected cable), it could take an unknown amount of time to reconnect.

cldodd, 2015-07-31:
Thank you. I will try these suggestions and see how things go. Chris.

closed 2015-07-31 by cldodd

ruscito, 2015-08-03:
Chris and Lynn,
After reading your posts I decided to make some changes to the lib to recovery the connection. I added a couple of exceptions “CommError” (generated if there are issues when the library is doing “communication things” send, receive, open and close) and “DataError” (generated when the library is handling data.   
Unfortunately EIP/AB documentation isn’t that clear on this topic, and maybe Lynn could provide better info.  On my test I noticed that if the connection breaks for a small period of time, a “forward_close” is enough to reestablish communication, but is the break is longer (at least more than 5 min) the only way to reestablish communication is to close end open back the connection.

See the following for details. 
Chris let me know if this works.

    from pycomm.ab_comm.clx import Driver as ClxDriver
    from pycomm.ab_comm.clx import CommError as ClxCommError
    from pycomm.ab_comm.clx import DataError as ClxDataError
    from time import sleep
    
    if __name__ == '__main__':
    
        c = ClxDriver(True, 'ClxDriver.log')
        try:
            c.open('172.16.2.161')
            retry = False
            number_of_retry = 0
            while 1:
                try:
                    if retry:
                        retry = False
                        print "retry in 3 seconds"
                        sleep(5)
                        c.forward_close()
                        if number_of_retry >= 5:
                            c.close()
                            c.open('172.16.2.161')
                            number_of_retry = 0
    
                    print(c.read_tag(['ControlWord']))
                    print(c.read_tag(['parts', 'ControlWord', 'Counts']))
                    print(c.write_tag([('Counts', 3424234234, 'INT'), ('ControlWord', 30, 'DINT'), ('parts', 40, 'DINT')]))
                    sleep(1)
    
                except ClxCommError as e:
                    retry = True
    
                except ClxDataError as e:
                    print (e)
    
            c.close()
    
        except ClxCommError as e:
            print (e)

cldodd, 2015-08-04:
Ruscito/Lynn
Following on from Lynn's suggestion of keeping the socket alive between data polls (ping) worked well, seems to recover from long & short disconnect periods.
Ruscito - Interesting, implementing your sample code to a CLX re-established a broken socket connection as long as the disconnect is between 1-3 retry periods (3 seconds). After this time it throws some random tag values then enters into an endless 3 second retry loop. A restart of the script obviously re-establishes immediately.  ....Chris

lynnlinse, 2015-08-04:
Lynn strongly suggests that during any error, just tear down and start with a new socket. This will be most predictable, and the least new code to write & test.

The problem is that even within AB's product lines, support for the 'purity' of the spec varies. For example, the ControlLogix is the poster-child for graceful perfection - actually using ForwardClose, so saying 'please' and 'thank you', etc. In contrast some of the older MicroLogix/SLC5 just hang up the socket. So I would worry that any clever (& spec-legal) attempt to heal a connection might suffer from unpredictability based on model & FW age ... and that's not even thinking about all the third-party I/O blocks supporting EIP, which might do something else!

ruscito, 2015-08-04:
Lynn that's a good point; bring more code in (and potentially bugs) to recover a socket is insane :)

Thanks to both of you for this conversation