ruscito/pycomm #5: Using pycomm to read Micro800

patrickjmcd, 2015-09-30:
Is there a way to use this module to read from a Micro800 PLC?

lynnlinse, 2015-10-01:
Is there a particular reason this doesn't work with an Ethernet-enabled u800?

I've never used an AB u800, but I'd assume they work much like the older uLogix, except I assume they are moving to add richer true CIP object support ... but still, if they expect a PLC5 or SLC5 to talk to it, they will still need the old PCCC objects supported. Ab learned a bad lesson when they first introduced 'Ethernet/IP', meaning ControlLogix and PLC3/PLC5/SLC5 could NOT work together. Bad news in marketing terms! The fix was to force the older PLC to add Ethernet/IP and the PCCC objects, plus CLogix gained the PCCC vendor objects.

- Lynn A Linse, Cradlepoint.com

ruscito, 2015-10-01:
Micro800 is a different "beast". AB has a different tech group that handle this PLC. The programming software is free (Workbench Software) and the communication is Modbus RTU based not EIP.  There are other projects on github that can help (check out pyModbusTCP)

Thanks,
Ago

closed 2015-10-02 by ruscito

lynnlinse, 2015-10-02:
That they use Modbus is interesting - but I'd wonder if they even download programs by Modbus? They could - SE/Modicon does (using proprietary codes). I should probably get one to learn more.

patrickjmcd, 2015-12-15:
I was able to get reading from a Micro820 to work by removing the lines 

    pack_usint(self.attribs['backplane']),
    pack_usint(self.attribs['cpu slot']),

from the forward_open and forward_close functions in cip\cip_base.py.

ruscito, 2015-12-17:
Could you run the same test but with the 'backplane' and 'cpu slot' set to zero? Unfortunately I don't have a MICRO 820 available.
Thanks for the help.

patrickjmcd, 2015-12-17:
I forgot to add that I changed the CONNECTION_SIZE['Backplane'] to CONNECTION_SIZE['Direct Network'] as well.

When I set the backplane and CPU Slot to 0 with 'Direct Network' connection size, I get the CIP Error "Too much data"

When I set the backplane and CPU Slot to 0 with 'Backplane' connection size, I get the CIP Error "Port Not Available"

patrickjmcd, 2015-12-17:
Here are some wireshark captures for each of the three scenarios

[Pycomm.zip](https://github.com/ruscito/pycomm/files/65700/Pycomm.zip)