ruscito/pycomm #6: Re: pycomm.ab_comm.clx & Strings

micksta, 2016-01-02:
Hi ruscito

On L plates here with python and github, got this up and running in no time. Firstly nice job straight up!, big thanks.

I have a query - I'm reading/writing DINT, REAL, DWORD no worries but bombing out reading STRINGS. Any thoughts

kindRegards
Mick

lynnlinse, 2016-01-02:
Strings are a really gnarly mess. Some other PLC hide strings in a simple INT file. Others (SLC5-ish) force the string to be a fixed size, starting with a 16-bit INT length and (I think) null or space padded. Others (ControlLogic-ish) will also have the same fixed size, but the length is 32-bit INT. If you use a 'tagged' sub-command, then you'll need to place the correct TYPE as expected by the PLC.

I'd run RSLogix or an OPC server to read a, run WireShark & see how the string is moved in your case. (My CIP experience is geared largely on old SLC/PLC3/MicroLogix stuff - not newer things like Micros)

lynnlinse, 2016-01-02:
I should add - I seem to remember the fixed size is 82 chars not including the 'length'. But I am traveling and without access to my old files to check. So to read 2 strings on a SLC5 would require 2 84-bit objects. Also, do watch the 'data TYPE field', as an older MicroLogix will require the correct data format, but it is still called a INT type - why it's safest to just use Wireshark to confirm what YOUR specific models want to see.

micksta, 2016-01-03:
Understood, thanks for the response. 
Great code by the way.... M

closed 2016-01-03 by micksta

ruscito, 2016-01-04:
Hi Mick,
thanks for your feedbacks. I didn't really tried to handle strings (mostly
because in the automation/control is not so common and because AB limited
document). As Lynn suggested WireShark will be very useful in this
case. I'm still trying to overcome the past holidays food fests, but as
soon as I have some time I will try to look into it.

Thanks,
Ago