ruscito/pycomm #8: Multiple Connections

PBrn46, 2016-06-15:
I am using this library to periodically read tag values. If I have more than one instance of the library running, one of the instances seem to not be able to read if they read the tag at the same time. I'm wondering whether this has anything to do with unique connection ID's or something like that? I have tried this with a single client or with different devices.

Also, this package has been very useful! Thank you for making this.

ruscito, 2016-06-15:
Hey Boris,
you can try to change the CID attribute to something else. The default
value is \x27\x04\x19\x71 but you can change it at run time before create
the connection (just keep it in 4 bytes length)

Which PLC are you using?

thanks,
Ago

PBrn46, 2016-06-15:
Great,  I will try that. I typically use 1769-L30ER or similar.

PBrn46, 2016-06-24:
I can confirm that changing the CID attribute did work. I have put a random generator for my last two bytes for the CID to ensure uniqueness. Thanks for your help!

ruscito, 2016-06-25:
Great!

closed 2016-06-25 by ruscito

themeanbean, 2017-02-27:
Want to just say awesome library! I'm currently using it my dissertation and its been a great help.
I'm interfacing with an AB SLC 5/05 CPU.
In regards to multiple connections changing the CID didn't work but changing the **VSN** did. 
I simply generate a random 4 byte string using os.urandom(4).
Thought I'd post just in case anyone else has this issue.

ruscito, 2017-03-05:
Thanks for the info. I added this functionality to the open method. Now the method open generate a random CID and VSN

AndrewStuhr, 2020-06-25:
I'm not sure why, but I need to manually change the VSN in order to allow multiple connections.  If I don't, it breaks the oldest connection.  I'm changing the VSN like this:

c = SlcDriver()
c.attribs['vsn'] = '{0}{1}{2}{3}'.format(chr(1),chr(2),chr(3),chr(5))

ruscito, 2020-06-25:
That's strange because the open method normally generate a new CID and a new random VSN. I will look in to but for now the way you change VSN is correct