ruscito/pycomm #11: Version Numbers

rkrupnek, 2016-08-09:
I'm not sure if this is an "issue" or not, but the modules have __version__ set at 0.1 and 0.2  It threw me off when I thought it should have been 1.0.8  

I'm just starting to play around with this a bit.  Thanks for the great work so far!

ruscito, 2016-08-10:
You are right the version should be 1.0.8. Where did you read revision 0.1 and 0.2?

rkrupnek, 2016-08-10:
I first noticed it by trying the sample code where you print the pycomm.ab_comm.clx.Driver version.

It looks like it shows up as 0.2 in cip_base.py and clx.py

slc.py has the version as 0.1

ruscito, 2016-09-10:
hey rkrupnek,
version 0.2 and 0.1 refer to the CLX and SLC modules  while 1.0.8 is the revision for pycomm. Pycomm will include also other protocols

closed 2016-09-10 by ruscito