ruscito/pycomm #12: Write_XXXX function parameters

rkrupnek, 2016-08-15:
The pycomm.ab_comm.clx.Driver has the default order of parameters (value & data type) switched for write_tag vs write_array.  It seems like this would be easier to keep straight if they were the same.  However, I may be all wet on this, and you might have a very good reason for doing it that way.

def write_tag(self, tag, **value=None, typ=None**):
    ...
vs
def write_array(self, tag, **data_type, values,** raw=False):
    ...

ruscito, 2016-09-10:
That's a good catch. There is now  good reason beside the method signature, and you are right it should be consistent across the other methods. However,  if I change it now there will be a compatibility issue with who using the library now. That's not an easy call lets put it on [StrawPoll ](http://www.strawpoll.me/11198665 )and see the feedback

ruscito, 2016-11-12:
Following the poll result I changed the order for the parameter

closed 2016-11-12 by ruscito