ruscito/pycomm #13: Unable to establish session

nehasahlawat, 2016-09-16:
Hi,

I am a newbie to both networking protocols as well as PLCs. I am trying to connect to a CompactLogix PLC  by providing the IP address as an argument for c.open('IP'). The issue is that the PLC I am supposed to access is on one of the slots, and I am unable to figure out how to enter this slot along with the IP address. Kindly help

patrickjmcd, 2016-09-16:
Which CompactLogix PLC are you using?

nehasahlawat, 2016-09-16:
It is ControlLogix 1756-L83 (not CompactLogix, sorry about that)

ruscito, 2016-09-16:
Can you elaborate a little more  how you are connected to the plc?

nehasahlawat, 2016-09-16:
The ControlLogix system is hosted on a given IP address, which I am using to open the socket. 1756-L83 plc is connected to this system on slot 5. Thus, in order to read or write tags, I need to access 5th slot from the open socket. I am using RSLinx gateway to create tags in PLC through Studio 5000.  My end goal is to be able to read that tag value through a TCP socket using the Read/Write services in your library.

ruscito, 2016-09-16:
I guess you already to test if you can ping the address successfully?

nehasahlawat, 2016-09-16:
I can ping the address successfully, but I am not able to read the tag value.  Attached is a s
[image]
creenshot

ruscito, 2016-09-16:
Can you attach the log file

Agostino

nehasahlawat, 2016-09-16:
[ClxLog.docx](https://github.com/ruscito/pycomm/files/477691/ClxLog.docx)

ruscito, 2016-09-16:
Looking at the log file there is no problem with the connection. Are you sure the tag you are trying to read  reading exists :)

nehasahlawat, 2016-09-16:
Hi, Yes the tag exists, but let me double check. Its only my first time using Studio5000.

But, how do I guarantee that the values being received are from the PLC on slot 5. Thanks a lot for all the help :)

ruscito, 2016-09-16:
The IP address is assigned to that PLC the location in the rack doesn't
really matter. I don't think you have more than one CU on the rack. If so
that will be the first time I see it. From the screenshot looks like you
get two reading 44818 and 0.2

Ago

nehasahlawat, 2016-09-16:
Yes, they are port and version values. If the IP address and slot doesn't really matter, then I assume its all about the tag. So, if the tag exists, I should be able to fetch and read it. Thanks for the help. Let me recreate the tags and try again.

nehasahlawat, 2016-09-19:
Hi,

So I have a tag called "MFETestTag" and have given it a string value "PLC" through studio 5000. Now through the example test code you have provided, I am trying to read that tag, like below

[image]

This is what I get when I run the program.
[image]

nehasahlawat, 2016-09-19:
I am also attaching the log file.

[log.docx](https://github.com/ruscito/pycomm/files/480757/log.docx)

Please let me know if I am doing anything wrong here, it will help me understand better.

nehasahlawat, 2016-09-19:
One more thing, if I replace {'MFETestTag') with (['MFETestTag']), I get the following output

[image]

[image]

ruscito, 2016-09-19:
STRING data type has not been implemented yet, so that's why it fails on
reading or writing the string, but the integer should be fine.

"read_tag" return none when the tag dose not exist, make sure the tag is
defined in the Controller Tags "folder". Tags defined outside this location
are not visible across the network.

Again looking at the log file I don't see any connection issue, the CPU
can't find the tag you are asking for.

patrickjmcd, 2016-09-19:
I'm pretty sure the library assumes that the CPU is in Slot 0 of the chassis for a ControlLogix processor.

from cip_base.py, line 500:

`self.attribs = {'context': '_pycomm_', 'protocol version': 1, 'rpi': 5000, 'port': 0xAF12, 'timeout': 10,
                        'backplane': 1, 'cpu slot': 0, 'option': 0, 'cid': '\x27\x04\x19\x71', 'csn': '\x27\x04',
                        'vid': '\x09\x10', 'vsn': '\x09\x10\x19\x71', 'name': 'Base', 'ip address': None}`

and then line 700:
`forward_open_msg[20:3] = [
                CONNECTION_SIZE['Backplane'],
                pack_usint(self.attribs['backplane']),
                pack_usint(self.attribs['cpu slot'])
            ]`

nehasahlawat, 2016-09-19:
Thanks! @patrickjmcd This helped. I am now able to read and write tag values. I went and changed the cpu slot in cip_base.py

@ruscito Thanks for letting me know, I will not use String for testing, which I have been doing all along.

Thanks for the help guys!

ruscito, 2016-09-19:
Yes the library assume the CPU is in slot zero, but as you pointed can be changed. I guess in the thread of messages I didn't read the cpu was on slot 5

Good catch Patrick :) 

@nehasahlawat I'm working on the string data type. A little more complicated because there is no official documentation, at the moment I'm able to read but not to write.

nehasahlawat, 2016-09-19:
Thanks ! Will wait for String support.

ruscito, 2016-09-19:
@nehasahlawat one last thing you don't need change the cip_base.py file in order to change the CPU slot.  You can change it at run time using the "attribs" dictionary; in your case will be 

 c.attribs['cpu slot'] = 5

Just make sure you do so before connecting

rkrupnek, 2016-09-19:
For what it's worth, I have come up with a method of using Pycomm for strings (at least on ControlLogix).  I don't have the code in front of me right now, but would be happy to share.  I'm not sure if this is the right place for that, but let me know if you are interested.

patrickjmcd, 2016-09-19:
I'd echo what @ruscito said, it's much more "pythonic" to use `c.attribs['cpu slot'] = 5` rather than editing a package's source.

A controllogix and compactlogix string tag consists of a .LEN DINT element and a .DATA array of 82 SINTs. You could probably write a function pretty quickly to read the .LEN element and then read that many elements of .DATA and parse them to chars.

nehasahlawat, 2016-09-19:
Yes @patrickjmcd  and @ruscito  I used the c.attribs to change cpu slot. This thread helped a lot and I am clearer on the functionalities in the library now. I will try to write my own function for reading String tags as well

@rkrupnek sure! that will be a lot of help. Will be happy to see what you have written and if it serves my use case as well.

ruscito, 2016-09-20:
yes @rkrupnek it will be nice if you can share your code. The approach I was taking is what @patrickjmcd  described but again if you have something done then share it :)

rkrupnek, 2016-09-20:
Okay, folks.  It's probably not the cleanest since it was just a quick test to see what I could do with pycomm, but here are the basics that seem to work well for me.  This could easily be improved by creating a send_string function.  I didn't do this yet, but if I were to do it, I would include the following features:  

1) Clear out the original string (all 82 characters).
2) Pack the string as done in the function below.
3) Send both the string data array and the length for the full update operation to be accepted correctly on the ControlLogix.

Notice that there is nothing special required to grab a string from the PLC.  This just gives the string as an array of tuples with the array index first, then the integer representation of the character.  So, if the string was "ABC", it would return [(0, 65), (1, 66), (2, 67)]  I didn't take the time to make a function to clean up the usage to give an actual string yet since I didn't really need it for my testing.  But this should be fairly trivial.

So, the actual code I'm using thus far is below.

`
def pack_string(to_pack = "01234567890"):
    trimmed = to_pack[0:82]
    conv = [ord(x) for x in trimmed]
    return conv
    packed = list(zip(range(len(conv)), conv))
    return(packed)
`

A simple usage example is shown below.
`    
print(c.read_array('Str1.DATA', 82))

test_string = "Urocyon Cinereoargenteus"
print(c.write_array('Str2.DATA', "SINT", pack_string(test_string)))
print(c.write_tag('Str2.LEN', len(test_string), "DINT"))
`

Is it madness?  Perhaps.  But hopefully, there is a method to my madness ;-)

ruscito, 2016-09-21:
That's not madness! I prefer to call it "software survival skills" :)
Looking at the log from wireshark there is much more going on, but your
code could be a fast workaround. I will try to add it to the lib in the
form of function, so that we can keep the same method signature when the
protocol get finalized.

At this point I will close this issue

thanks everybody for the effort put on this

closed 2016-09-21 by ruscito

nehasahlawat, 2016-09-21:
@ruscito  I used the method suggested by @patrickjmcd and am able to read string values too! Thanks for the help guys!