ruscito/pycomm #15: Problem writing arrays to ControlLogix PLC

twobitpunk, 2016-09-26:
Hi!

I am doing a project that requires reading and writing tags to Allen Bradley PLCs, so PyComm comes in very handy. I am having an issue writing arrays to the PLC however:

```
testarray = [12, 13, 14]
driver = clx()

if driver.open('192.168.2.111'): 
   driver.write_array('Name.Space.Tag', 'INT', testarray) 
   driver.read_array('Name.Space.Tag', 3) 
   driver.close() 
```

Running this, I get this rather cryptic error:

```
File "C:\Python27\lib\site-packages\pycomm\ab_comm\clx.py", line 453, in read_array
    addr_data=self._target_cid,
  File "C:\Python27\lib\site-packages\pycomm\cip\cip_base.py", line 591, in send_unit_data
    return self._check_reply()
  File "C:\Python27\lib\site-packages\pycomm\ab_comm\clx.py", line 337, in _check_reply
    raise DataError(e)
pycomm.cip.cip_base.DataError: unpack requires a string argument of length 2
```

I can do the write to the PLC, but the value in the PLC tag does not change. When I try to read it back, I get the error above.

I can read and write simple tags without any issues - but array operations seem to fail - can you point me in the direction of a fix or a workaraound for this?

Thank you!

Sincerely,

Lars Stoustrup

ruscito, 2016-09-27:
Lars, 
Could you try to move the array out of the UDT and execute the read and write again? Maybe there is a problem when the array is inside the structure. I don't have a PLC handy to run the test at the moment

twobitpunk, 2016-09-27:
Thanks a lot! I figured it out it was a naming thing, more or less:

Apparently, when a tag is present in the PLC but not made remotely accessible, the error returned by the PLC is different from the one you get if the tag is not present at all. Making the tag remotely accessible resolved the issue.

Thanks for taking the time to respond - and PyComm works great :-)

closed 2016-09-27 by twobitpunk