ruscito/pycomm #17: Strings

INTERCEPTOR27, 2016-11-01:
Is anyone having trouble retrieving strings from pycomm ?

rkrupnek, 2016-11-01:
ControlLogix strings can be a bit tricky.  pycomm does not really handle the STRING type as a true native type, but I have some code I posted earlier which does the job pretty well.  Actually, if anyone else who read the earlier stuff sees this, I have made some improvements.  It's just a couple short "add-on" functions for reading / writing strings with pycomm's existing functionality.

`def write_string(driver, string_tag, to_send = "01234567890"):
    # Create the actual tag names:
    dataname = ".".join((string_tag, "DATA"))
    lenname = ".".join((string_tag, "LEN"))

    # Start by clearing out the string array:
    clear = [0 for _ in range(82)]
    driver.write_array(dataname, "SINT", clear)
    driver.write_tag(lenname, 0, "DINT")

    trimmed = to_send[0:82] # trim to CLX limit
    conv = [ord(x) for x in trimmed]
    driver.write_array(dataname, "SINT", conv)
    driver.write_tag(lenname, len(to_send), "DINT")

def read_string(driver, string_tag):
    # Create the tag names we need to work with:
    dataname = ".".join((string_tag, "DATA"))
    lenname = ".".join((string_tag, "LEN"))

    # Read the data & parse the array.
    clxlen = driver.read_tag(lenname)
    arraydata = driver.read_array(dataname, clxlen[0])
    unzipped = zip(*arraydata)[1]  # we only want the 2nd element (actual data)
    letters = [chr(x) for x in unzipped]
    return "".join(letters)`

INTERCEPTOR27, 2016-11-01:
Thanks rkrupnek, I'll have a look at that and see if I can put it to use. Do you have any documentation on the Ethernet/IP standard?

rkrupnek, 2016-11-01:
Cool.  Hope you find it useful.  Ruscito has this posted on the front GitHub page.  I think there is a later version of it available as well: [http://literature.rockwellautomation.com/idc/groups/literature/documents/pm/1756-pm020_-en-p.pdf](http://literature.rockwellautomation.com/idc/groups/literature/documents/pm/1756-pm020_-en-p.pdf)

ruscito, 2017-04-23:
ok guys I added two temporary method to read and write string. I was holding off to do so because I wanted to do it right (using read/write fragment wrapped around a "Multiple Service Packet") but is taking longer than what I expected  so for the moment this two methods can do the trick.  @rkrupnek  thanks for

closed 2017-04-23 by ruscito