ruscito/pycomm #21: Can't write/read to arrays, but tags fine

sgchowar, 2017-03-24:
Im trying to read/write to arrays, I can do it with tags even of the same datatype. The array functions return nothing not even an error, as if the arrays didn't exist. 

import sys
from pycomm.ab_comm.clx import Driver as ClxDriver

x_array = [ ]

c= ClxDriver()

if c.open("192.168.3.144"):
      x_array = c.read_array("Test",32)
      for tag in x_array:
            print (tag)
      c.write_tag(X22,1,'BOOL')

c.close

As you can see im also writing to a tag and going on RsLogix i can see the 1 being written to tags X22. In the same tag table i've added the Test array, so everything should work.  If the X22 tag which was added in the same table after adding the test array, surely if writing to X22 works, the array should work also.

Thanks .

sgchowar, 2017-03-24:
I have managed to get something to work, but it seems to only read/write the arrays as arrays of dint/int etc, not as seperate bools. Does this therefore not support writing to an array of bools.

sgchowar, 2017-03-24:
Basically i have been able to write and read from the array of bools, but i cannot do it to individual bools in the array, instead the most amount of parts i can split the 128 bit array is into 4 parts, therefore viewing it as a 4 DINT array. Increasing the element count in the functions to 8 or 128 as i really want, doesn't return any output or error.

Is this a limitation with the library or something i am missing?

rkrupnek, 2017-03-24:
Here is how I am reading from a (64 element) boolean array using my little test program:

print(bin(c.read_tag('B')[0])[2:64])

It prints the result like this:  
111100001111

If you just straight-up read the content of the boolean array, you get:
(3855, 'DWORD')

The limitation is most likely due to a combination of the way Rockwell handles boolean arrays (as integers / blocks of bits) in the processor and perhaps the way Python handles this data.  It could probably be worked around.  I'll try to give it some thought.

ruscito, 2017-03-24:
I have run a test and look like the _parse_fragment function read an "unknown status" from the PLC. I will need to look at the log files to understand what's happening. Will keep you posted

sgchowar, 2017-03-27:
Thanks it would be great if you could get it working. 

rkrupnek, I used your method to read from an element in my array, but you cannot use this to write to a single element.

ruscito, 2017-03-29:
Ok so this is more of a Rockwell representation. In other words Rockwell see the array of bool as DWORD type for both performance and space reasons (I read this somewhere in the documentation). If you try to define an array of BOOL of dimension 3 Rockwell will automatically create an array of dimension 32 BOOLS, or if you try to define an array of dimension 33 Rockwell (RSLogix I should say)  will automatically create an array of 64.

The only way at the moment is to  use the read and write array function as DWORD size for example: 
I defined in my plc an array of 32 Boolean; now if I need set the first 3 bool to 1 i will have to write 15 in the whole DWORD . 

        w_array = []
        w_array.append(15)
        c.write_array("AGO_BOOL_TEST", w_array, "DWORD")

If i need to read the array I will need to pass the size in term of DWORD , so if I need to read the first 4 bit i will have to use size of 1 because this 4 bits resize in the first DWORD  

        r_array = c.read_array("AGO_BOOL_TEST", 1)

This will read the entire word. 

This is not the best solution because you will need to do some coding to extrapolate the bit needed. To be honest this code could be easily write inside both "read_array" and "write_array"  and I will try to do it but later because now I'm in the middle of a startup. 

Hopefully this help a bit
Ago