ruscito/pycomm #26: How can I do an "atomic" bit write using the write_tag()

gman8a, 2017-04-15:
See Wiki page for my "Read-Modify-Write" code.

When multiple clients are connected to a PLC,  and the "bit write" method is accomplished at the application layer, by doing  "Read-Modify-Write" operation, then there is a good possibility the PLC bit write register will become corrupted, as there are no lock semaphore across connected clients.

Thus the low-level-driver (LLD) needs to support the AB bit write.

Question: Does pycomm support device bit write ?

ruscito, 2017-04-15:
Not really sure what are you asking, but you can use "write_tag" to write a boolean @ #

gman8a, 2017-04-15:
see AB Control logic manual 

  URL: http://literature.rockwellautomation.com/idc/groups/literature/documents/pm/1756-pm020_-en-p.pdf

page 30

Read Modify Write Tag Service
The Read Modify Write Tag Service provides a means to modify Tag data with
individual bit resolution. Within ControlLogix, the Tag data is read, the logical
or modification masks, or both, are applied, and finally the data is written back to
the Tag. It can be used to modify a single bit within a Tag without disturbing
other data. Its purpose is similar to the PLC-5 style Read Modify Write PCCC
command described in Chapter 5

gman8a, 2017-04-15:
I see the Read Modify Write Tag Service (0x4e) was not implemented.

ruscito, 2017-04-18:
gman8a yes you are right Read Modify Tag Service is not implemented yet; I will take a look, but at the moment I'm focusing more to handle STRING datatype correctly

Thanks
Ago

closed 2017-08-08 by ruscito