ruscito/pycomm #27: self.forward_open()

shah-s, 2017-06-30:
Hey,
I am trying to run test_slc_only.py but I am getting an error in line
build_common_packet_format(DATA_ITEM['Unconnected'], ''.join(forward_open_msg),ADDRESS_ITEM['UCMM'],)):
Can you please help me in resolving it.

ruscito, 2017-06-30:
could you post the error message and the source code?

shah-s, 2017-07-05:
I am trying to run the exact code which you have posted with the changed IP address.

shah-s, 2017-07-05:
[image]

ruscito, 2017-07-05:
It looks like you didn't connect with the PLC

shah-s, 2017-07-05:
So in "cip_base.py" code in forward_open function do i need to change Connection parameter from default to PLC?

ruscito, 2017-07-05:
I don't think so. I tested the connection with different PLCs including SLC
and MicroLogix and I noticed that PLC always responded to the 'Default'
parameter.
Therefor before make any changes I would first make sure the PLC is
available (can you ping it, is the Ethernet port available with a
reasonable configuration, how do you go online with RSLogix?)

Ago

shah-s, 2017-07-05:
Yes, I am able to ping to the IP address where PLC is connected. 
So what can be the issue for this error?

ruscito, 2017-07-05:
Difficult to help without a log file. Delete the existing log file, execute
the script again and send or post the log

shah-s, 2017-07-05:
I did not understood which log file you are talking about. As i am using the exact code of yours for now. 
Can you please explain which log file you want me to execute again?
thanks

ruscito, 2017-07-05:
In the directory were you are executing the script there is a file "SlcDriver.log" which contains debug information and the data packets exchanged. You need to delete this file, and execute the python script again. Once you do so the log file will be recreated with the latest debug information

shah-s, 2017-07-05:
Here is the new log file
[image]

shah-s, 2017-07-05:
[image]

ruscito, 2017-07-05:
it looks like it fail during forward_open but i'm not sure from the image you posted. It is possible to get the entire log file

shah-s, 2017-07-07:
Should I send the images of log file?

shah-s, 2017-07-07:
here is the pdf version of the log file
[doc.pdf](https://github.com/ruscito/pycomm/files/1131078/doc.pdf)

ruscito, 2017-07-07:
As I suspected the forward_open failed. Which  PLC model are you trying to connect and what's the tag you are trying to read?

shah-s, 2017-07-07:
PLC model is Micro820 and Tag I am trying to read is DV

ruscito, 2017-07-07:
Harggg!  I should ask this question at the beginning. Micro models are totally different beast :) So they respond as "Out of Connections"  (0x0113 is the status I read from your log)  
Anyhow to work you need to set the parameter "direct_connection" to True  when you open the connection

c.open(172.16.2.167, direct_connection=True):

shah-s, 2017-07-07:
Thank you, it worked.
I am getting 22 as output, what does that signifies?

ruscito, 2017-07-07:
it should be the value of your tag. Make sure to import  "pycomm.ab_comm.clx"  and not  "pycomm.ab_comm.slc"

shah-s, 2017-07-07:
okay. thanks a lot for your help!

ruscito, 2017-07-07:
No problem

closed 2017-07-07 by ruscito

shah-s, 2017-07-07:
One last thing,
When i Interrupt the program i get the following as a result.
[image]
And can you explain me how to read data from log file?
[image]
 We will get the readings of PLC on log file right?

shah-s, 2017-07-07:
One last thing,
When i Interrupt the program i get the following as a result.
[image: image]
<https://user-images.githubusercontent.com/29803896/27967949-fde385c6-6312-11e7-839a-ce5abf158c76.png>
And can you explain me how to read data from log file?
[image: image]
<https://user-images.githubusercontent.com/29803896/27967980-2cf50a74-6313-11e7-9f0c-affd204db591.png>
We will get the readings of PLC on log file right?

ruscito, 2017-07-07:
For the first question: 
the reason you are getting the error is because you interrupted the process with ctrl-c without close the connection. The example you are using is not the best "strategy", what you really want is to exit the reading/writing loop and then call the "close" method to clean-up the connection with the PLC  (foward_close)
 
For the second question:
the log file is just debug purpose only; at the beginning in the example file we set the log level to "DEBUG", but if you set it to "WARNING"  then the log file will contain only errors or warnings. There is are lots of information in the log file and the majority is protocol related that wrap around the value you are trying to read

shah-s, 2017-07-07:
So how to read value from log file?

ruscito, 2017-07-07:
You don't! You read value returned by the read_tag() method and you write using the write_tag() method

shah-s, 2017-07-10:
Can you explain what each print statement signifies in "test_slc_only.py"
program.
I am trying to read data from PLC Micro820. It is connected with multiple
appliance so is there any way we can filter out data of particular
appliance?

shah-s, 2017-07-10:
How to solve the error of forward_close returned False
[image]

ruscito, 2017-07-10:
That's not an error. When you connect for the first time pycomm try to execute the foward_close to close any nonactive connection ( a connection that was not closed correctly could stay active reducing the number of resource ) In this case it returned false just because there wasn't opened connections

shah-s, 2017-07-11:
can you explain how u made readtags() in program and from where we can read the values returned by the readtag()

ruscito, 2017-07-11:
readtags() return the value, and you will need to write a program that put
the returned values  wherever you need it. You will need some programming
skills to do so

shah-s, 2017-07-11:
So the readtags() created by you in the programs would not help me? 
as my projects matches exactly with your example. I am trying to read the value from PLC from the remote desktop.

patrickjmcd, 2017-07-13:
If you're trying using a micro820, you need to use the other example "test_clx_comm.py". As @ruscito mentioned, you'll need to make sure you're enabling the direct connection by using 
`c.open(172.16.2.167, direct_connection=True)`

 In that example you'll see the line 
`print(c.read_tag(['ControlWord']))`

This should print out the value returned from c.read_tag() which is a tuple of (value, tag type). 

Storing the value of read_tags() will allow you to access the 0-th member of the tuple to get the value.

shah-s, 2017-07-13:
I got the error "A session need to be registered before to call forward
open"

patrickjmcd, 2017-07-13:
It would really help us to see the exact code you're using.

shah-s, 2017-07-13:
For now using the exact code which is there in "test_clx_comm.py"
As I am in learning phase.

patrickjmcd, 2017-07-14:
You’re changing the IP address to reflect the IP address of your PLC,
right? Please post the EXACT code you’re using.

> It would really help us to see the exact code you're using.
>
> <https://github.com/ruscito/pycomm/issues/27#issuecomment-315133785>, or
mute
> the thread
> <
https://github.com/

Saurabh Shah
*Information Technology Intern, Red Meters*
*Direct | *+1 407-337-0110
redmeters.com <http://www.redmeters.com/> *|* s.shah@redmeters.com
*The RM Series Technology & Specifications
<http://www.redmeters.com/technology/>*

<https://github.com/ruscito/pycomm/issues/27#issuecomment-315134564>, or mute
the thread

shah-s, 2017-07-14:
[image]
I am using this code.
Thank you

shah-s, 2017-07-17:
> AcbFePZlAmTdrK85Z3pK1CBN9u7HzHk2ks5sNkhAgaJpZM4OK8i3
> *Direct | *+1 407-337-0110 <(407)%20337-0110>

Can you see the code and let me know if i need to make any changes in that
or not.

patrickjmcd, 2017-07-17:
Do you have a tag named “ControlWord” in your Micro800?

What model of Micro800 are you using? 820? 850?

shah-s, 2017-07-17:
Micro 820

patrickjmcd, 2017-07-17:
Please post the whole error message you get, including the command used to run the script

shah-s, 2017-07-17:
[image]

patrickjmcd, 2017-07-17:
Can you remove the while loop, so it just runs once and post that output?

shah-s, 2017-07-17:
[image]

shah-s, 2017-07-17:
Do i need to make any changes in Forward_open() ?

shah-s, 2017-07-18:
How can I solve this error?

ruscito, 2017-07-18:
shah-s I do you have a tag **"ControlWord"** in your PLC program?

ruscito, 2017-07-18:
That's not in the code you posted!! Please post  your code as is and a clean log file because the info you are giving are confusing and don't help

shah-s, 2017-07-18:
[image: Inline image 1]

[image: Inline image 2]

[image: Inline image 3]
this is the code and the logfile generated

patrickjmcd, 2017-07-18:
It would be really helpful if you posted the actual code, not just a screenshot of it. Plus, your screenshots didn't work.

surrounding text with a backticks (`) to format it as code

`like this`

shah-s, 2017-07-18:
I have attached the txt file of the code i am using, hope this helps you.

from pycomm.ab_comm.clx import Driver as ClxDriver
import logging

from time import sleep

if __name__ == '__main__':

    logging.basicConfig(
        filename="ClxDriver.log",
        format="%(levelname)-10s %(asctime)s %(message)s",
        level=logging.DEBUG
    )
    c = ClxDriver()

    print c['port']
    print c.__version__

    if c.open('192.168.150.60', direct_connection=True):
            try:
                print(c.read_tag(['DV']))
                #print(c.read_tag(['parts', 'ControlWord', 'Counts']))

                #print(c.write_tag('Counts', -26, 'INT'))
                #print(c.write_tag(('Counts', 26, 'INT')))
                #print(c.write_tag([('Counts', 26, 'INT')]))
                #print(c.write_tag([('Counts', -26, 'INT'), ('ControlWord', -30, 'DINT'), ('parts', 31, 'DINT')]))
                sleep(1)
            except Exception as e:
                c.close()
                print e
                pass

        # To read an array
        r_array = c.read_array("TotalCount", 1750)
        for tag in r_array:
            print (tag)

        c.close(

shah-s, 2017-07-19:
What changes should I make in the code? Do I also need to make changes in
cip_base.py?

On Tue, Jul 18, 2017 at 1:29 PM, Saurabh Shah <s.shah@redmeters.com> wrote:

> I have attached the txt file of the code i am using, hope this helps you.
>

-- 

Saurabh Shah
*Information Technology Intern, Red Meters*
*Direct  |  *+1 407-337-0110
redmeters.com <http://www.redmeters.com/> *|* s.shah@redmeters.com
*The RM Series Technology & Specifications
<http://www.redmeters.com/technology/>*

ruscito, 2017-07-19:
shah-s I'm still waiting for the clean log file

shah-s, 2017-07-19:
Here is the recent log file.

ruscito, 2017-07-19:
can't see any log file

shah-s, 2017-07-20:
I have attached log file. Sorry I forgot to attach it last time.

[log.txt](https://github.com/ruscito/pycomm/files/1162256/log.txt)

ruscito, 2017-07-20:
as you can see from the picture below
 
[image]

the error is "Service not supported" which practically means that the controller dose not recognize the "ReadTag" request  (service 4C). Unfortunately I don't have a Micro820 to play with so I can't help; however   the "ReadTag" request looks good, so  my best guess is with the controller configuration or maybe the tag not been exposed

shah-s, 2017-07-21:
So any advice to solve this error?

[2 more replies]