ruscito/pycomm #33: CPU Slot static variable

GreybeardFM, 2017-09-01:
CIP_Base has a static attribute, 'cpu slot', set to 0.  In many cases, the PLC you wish to communicate with is _not_ in slot 0.  

Adding an optional parameter to your open:
def open(self, ip_address, plcslot = 0, direct_connection=False):

... and then overwriting that static attribute a few lines into the routine...

self.attribs['cpu slot'] = plcslot

seems to correct the issue.  No changes required to working code, but, if you need to specify a different slot, you can do so when you open the connection:

if c.open('192.168.20.50', 3):

chris@midkiff.us

ruscito, 2017-09-10:
Hey Chris,
sorry to replay so late. What you proposed is a good idea and I already though about it; however, I didn't took that route though,  to keep things simple. 
That is also the purpose of having the attribs array, there are many more parameters that can be changed, and if we put plcslot, why not to put others?
So again to keep thing simple I decided to have all those parameters treated as attributes, that can be changed before open the connection. Hopefully this make more sense to you

Thanks
Ago

GreybeardFM, 2017-09-11:
Ago,
Thanks for the response.  I'm probably going to continue using my modified version of your library.  I understand your reasoning, but for the tasks that I'm working on, this works pretty well.  Aside from this issue, my testing, both from Windows and Linux Python 2.7, against several different model PLC's, at several different firmware levels, the library seems quite stable, and fast.  Not as flexible as the full OPC implementations I'm used to using, but lighter weight, quicker to implement basic read/write apps, and quite reliable.  Good, easy to understand code.  Took me about 5 minutes to find and make the change I proposed.

closed 2017-09-11 by GreybeardFM