ruscito/pycomm #37: Problem with Micrologix1100

A-Badkoubeh, 2018-01-20:
I tried the example "test_slc_only.py", but I am getting this error "Traceback (most recent call last):
  File "test_slc_only.py", line 13, in <module>
    if c.open('192.168.0.9'):
  File "/usr/local/lib/python2.7/dist-packages/pycomm/cip/cip_base.py", line 843                             , in open
    raise CommError(e)
pycomm.cip.cip_base.CommError: Socket timeout during connection."

I wonder if you tried this example on Micrologix 1100?

ruscito, 2018-01-20:
Can you ping that IP address? Usually you get that error when address is not reachable

A-Badkoubeh, 2018-01-20:
That is what I am getting when I ping the IP address:
"PING 192.168.0.9 (192.168.0.9) 56(84) bytes of data."

A-Badkoubeh, 2018-01-20:
Here what I get when I ping the IP: "PING 192.168.0.9 (192.168.0.9) 56(84)
bytes of data."

ruscito, 2018-01-20:
That's the command you sent. What's the replay? You should see something like 

Reply from 98.138.252.38: bytes=32 time=34ms TTL=52

A-Badkoubeh, 2018-01-20:
Sorry part of that didn't copied!!! Yes I am getting: 
PING 192.168.0.9 (192.168.0.9) 56(84) bytes of data.
64 bytes from 192.168.0.9: icmp_seq=1 ttl=128 time=5.56 ms
64 bytes from 192.168.0.9: icmp_seq=2 ttl=128 time=1.82 ms
64 bytes from 192.168.0.9: icmp_seq=3 ttl=128 time=2.40 ms
64 bytes from 192.168.0.9: icmp_seq=4 ttl=128 time=1.35 ms
64 bytes from 192.168.0.9: icmp_seq=5 ttl=128 time=2.15 ms
64 bytes from 192.168.0.9: icmp_seq=6 ttl=128 time=0.799 ms
64 bytes from 192.168.0.9: icmp_seq=7 ttl=128 time=1.28 ms
64 bytes from 192.168.0.9: icmp_seq=8 ttl=128 time=1.25 ms
64 bytes from 192.168.0.9: icmp_seq=9 ttl=128 time=1.45 ms
64 bytes from 192.168.0.9: icmp_seq=10 ttl=128 time=2.32 ms
64 bytes from 192.168.0.9: icmp_seq=11 ttl=128 time=2.37 ms

A-Badkoubeh, 2018-01-20:
I fixed the problem with the Ip. Now I'm getting this error:
"Error parsing the tag passed to read_tag(N:1/5,1)"

ruscito, 2018-01-20:
Hmmm strange, I just did a project with a 1100 without problem! How did you install the library?

ruscito, 2018-01-20:
please review the examples provided that's not the right way to make the call. You should yuse something like  

`read_tag('N1:5',`` 1)`

A-Badkoubeh, 2018-01-20:
1. here the complete error I am getting:
"Error parsing the tag passed to read_tag(N:0,1)
Error parsing the tag passed to read_tag(N:0,1)
Error parsing the tag passed to read_tag(N:0,1)
^CTraceback (most recent call last):
  File "test_slc_only.py", line 17, in <module>
    print c.read_tag('N:0')
  File "/usr/local/lib/python2.7/dist-packages/pycomm/ab_comm/slc.py", line
352,
      in read_tag
    logger.warning(self._status)
  File "/usr/lib/python2.7/logging/__init__.py", line 1172, in warning
    self._log(WARNING, msg, args, **kwargs)
  File "/usr/lib/python2.7/logging/__init__.py", line 1279, in _log
    self.handle(record)
  File "/usr/lib/python2.7/logging/__init__.py", line 1289, in handle
    self.callHandlers(record)
  File "/usr/lib/python2.7/logging/__init__.py", line 1329, in callHandlers
    hdlr.handle(record)
  File "/usr/lib/python2.7/logging/__init__.py", line 757, in handle
    self.emit(record)
  File "/usr/lib/python2.7/logging/__init__.py", line 950, in emit
    StreamHandler.emit(self, record)
  File "/usr/lib/python2.7/logging/__init__.py", line 859, in emit
    msg = self.format(record)
  File "/usr/lib/python2.7/logging/__init__.py", line 732, in format
    return fmt.format(record)
  File "/usr/lib/python2.7/logging/__init__.py", line 471, in format
    record.message = record.getMessage()
  File "/usr/lib/python2.7/logging/__init__.py", line 329, in getMessage
    if not isinstance(msg, basestring):
KeyboardInterrupt"

2.  How did you install the library?  I followed the steps in this page:
https://github.com/ruscito/pycomm

A

ruscito, 2018-01-20:
read_tag('N1:5',1)

A-Badkoubeh, 2018-01-20:
Cool Buddy! silly mistake!!! I changed it to " print c.read_tag('N7:1')".

I am successfully reading the value of this tag, N7:1.

I should be able to do the write similarly. correct?

Thank you very much.

ruscito, 2018-01-20:
no problem bud. It should be same for the write, just check the example from the main page. I think I covered all the possible option for pycomm. I'm out for now because is beer Friday :) Please close the issue if you have not more issues

ciao

A-Badkoubeh, 2018-01-20:
Works nicely! Thank you very much Agostion, ruscito. 
The example from the main page covers all the possible options for Pycomm! 
Cheers!

closed 2018-01-20 by A-Badkoubeh