ruscito/pycomm #38: What is the sampling rate here?

A-Badkoubeh, 2018-02-10:
On what update rate Pycomm read/writes data from PLC?

tyoungNIO, 2018-04-05:
I don't believe that explicit messaging is viable at that speed, so no, pycomm probably won't work for you.

ruscito, 2018-04-06:
I'm not sure I understood your question. Pycomm is just a lib your program speed (compiler as well as computer performance) dictates timing for the read call. However as @tyoungNIO  pointed out explicit messaging limitation could prevent that speed

closed 2018-04-06 by ruscito

A-Badkoubeh, 2018-04-06:
I agree with Agostino. My tests and reaseach shows Ethernet does nnot have
a deterministic speed. It gives you certain bandwidth and the speed is
determined by the performance of the system.
For Explicit massaging, I set the timer on 1MSec. So, that should not be a
limiting factor.