ruscito/pycomm #39: "Timeed out" Issue

A-Badkoubeh, 2018-02-10:
I am using the example for Micrologix PLC. After a few minutes, I am getting "timed out". Any ideas what potentially can cause this "timed out"?

lynnlinse, 2018-02-11:
You didn't give much info, but keep in mind that if you are 'routing' to get to the PLC, for example from your desk or control room to a factory floor, then a TCP/IP socket will likely 'die' very ungracefully if it is idle for 5 minutes. This is a router/firewall thing, unrelated to CIP timeouts. So if you are polling once a minute, then this isn't the problem, but if you have been idle, then try to talk, the socket might now be broken. You'd need to enable 'TCP keepalive' on the socket.

davisjosephh, 2018-05-24:
If you get those kinds of speeds let me know what you did. As is, I struggle to keep my reads under 1s.

Nsp2121, 2018-08-08:
I have the same issue with this
for reading one tag it takes about 60-70 ms and i am reading 18 tags in a while loop