ruscito/pycomm #41: Trying to push some commits for Python 3.6 compatibility.

jmbinette, 2018-02-22:
Not sure what I am doinhg wrong

patrickjmcd, 2018-02-22:
It looks like you created a PR for it. It just needs to be accepted before it can be merged into the master branch. I did a quick test of the code and it looks like reading/writing REALS on a CompactLogix and on a Micro800 using the direct_connection=True parameter of c.open().

There are a few types implemented by the Micro800 that do not work (UDINT, for example) but I will work to test which ones don't work and implement fixes. These just seem to be missing from the PACK_DATA_FUNCTION and UNPACK_DATA_FUNCTION structures and associated functions.

patrickjmcd, 2018-02-22:
Your code, however, does not seem to be backwards compatible with python2