ruscito/pycomm #42: Error when reading string

smarth23, 2018-05-21:
Error I am getting,
pycomm.cip.cip_base.DataError: local variable 'data_type' referenced before assignment.

I don't see it where the variable 'data_type' is referred before a it is assigned. Also the code throws the error above once I run it, but works fine when I try to run it again.

Alex-Atkinson, 2018-11-06:
@smarth23,
pycomm does not include functionality for reading strings in its default read_tag function. There is some functionality for reading a string shown in the readme file, but when checking the clx.py file there is no matching function present. Issue #17 shows some work done in supporting a string read, but this doesn't appear to have gone anywhere. Might be time for somebody to create a branch and look into this.

Alex-Atkinson, 2018-11-06:
Did a little more checking on this today after getting ready to add this functionality to a new fork and found that the read_string function is present in GitHub but not in the PyPi version. Looks like the new version just needs to be pushed over to PyPi. According to the [pycomm page](https://pypi.org/project/pycomm/) there has not been an update to the PyPi version since November 1, 2015.