ruscito/pycomm #46: Connecting to Productivity PLC via pycomm

Hux123, 2018-07-19:
I'm trying to connect to a productivity plc with pycomm using the TCP/IP Protocol using the following code:

from pycomm.ab_comm.slc import Driver as SlcDriver
c = SlcDriver()
c.open("[MY IP ADDRESS]")

I keep getting "[Errno 111] Connection Refused".

My first question is whether or not I can use pycomm for the productivity PLC. Is the pycomm library only good for Alley Bradley PLC's?
If it is only good for AB PLC's, is there another module such as cpppo or pymodbus that would be better suited for this application?

ruscito, 2018-07-20:
Is productivity PLC Ethernet/IP based?

Hux123, 2018-07-20:
Yes it is ethernet/up based. I am currently able to ping the plc and I have solved the "connection refused" problem, however I am still unable to access any of the tags for reading/writing.