ruscito/pycomm #47: Communication with Micro 850 Plc

muthunsr, 2018-07-26:
I am trying to communicate with micro 850 PLC and this is not happening it gives a error message

muthunsr, 2018-07-27:
Thanks for replay patrick

I removed the line as  per your previous command. ok i will check

> 
> "I was able to get reading from a Micro820 to work by removing the lines
>  
>  pack_usint(self.attribs['backplane']),
>  pack_usint(self.attribs['cpu slot']), 
> 
> from the forward_open and forward_close functions in cip\cip_base.py

```
from pycomm.ab_comm.clx import Driver as ClxDriver
import logging

if __name__ == '__main__':
    logging.basicConfig(
        filename="ClxDriver.log",
        format="%(levelname)-10s %(asctime)s %(message)s",
        level=logging.DEBUG
    )
    c = ClxDriver()

    if c.open('192.168.1.35'):
        a = c.read_tag(['Value_A'])
        
        print(a)
        

        c.close
```

patrickjmcd, 2018-07-27:
You don't need to modify the package, you just need to add the `direct_connection=True` to your `open(...)` call.

Like this:

```python

from pycomm.ab_comm.clx import Driver as ClxDriver
import logging

if __name__ == '__main__':
    logging.basicConfig(
        filename="ClxDriver.log",
        format="%(levelname)-10s %(asctime)s %(message)s",
        level=logging.DEBUG
    )
    c = ClxDriver()

    if c.open('192.168.1.35', direct_connection=True):
        a = c.read_tag('Value_A')
        
        print(a)
        

        c.close

```

muthunsr, 2018-07-28:
Thanks For the replay 
After including the " direct_connection=True " i got an error please look 

coding used for testing 

```
if __name__ == '__main__':
    logging.basicConfig(
        filename="ClxDriver.log",
        format="%(levelname)-10s %(asctime)s %(message)s",
        level=logging.DEBUG
    )
    c = ClxDriver()

    if c.open('192.168.1.35', direct_connection=True):
        a = c.read_tag('Value_A')
        print(a)
        c.close()
```
Error after executing this code 

```
Traceback (most recent call last):
  File "Micro_800_Comm.py", line 13, in <module>
    if c.open('192.168.1.35', direct_connection=True):
TypeError: open() got an unexpected keyword argument 'direct_connection'
```

patrickjmcd, 2018-07-28:
You'll need to clone the GitHub repo and use this version, not the version that is on pypi

muthunsr, 2018-08-01:
Thanks Now the micro 850 PLC is Communicating

closed 2018-08-01 by muthunsr