ruscito/pycomm #62: Micrologix 1200

nah2015, 2019-11-06:
So I am able to successfully read/write tags to a Micrologix 1400 over Ethernet/IP, but now I am in a situation where I need to read/write tags on a Micrologix 1200.

Can this work using a USB to Allen Bradley serial cable typically used on the PLC?

I would plug the USB end of that cable into a USB type a female to RJ45 male adapter and then plug it into my ethernet.

I'm guessing there is no chance this will work but its the only option for reading/writing tags with this library on a MicroLogix 1200 that I can think of.

Perhaps I'm just not understanding things though.  Is it not possible at all to work on a MicroLogix 1200?

ruscito, 2019-11-06:
Micrologix 1200 do not supports Ethernet/IP so I'm not sure even using the converter you will be able to communicate. Maybe you could use [https://github.com/riptideio/pymodbus](url) and communicate via modbus

nah2015, 2019-11-07:
I am going to attempt to use that with my AB serial to usb cable.  I have no idea what I am doing as I have never even heard of the term "coil" but I'll give it a try haha

Thanks for the quick response

nah2015, 2019-11-07:
Sorry to keep bothering you about this.  I apologize but I am not very knowledgable about PLCs so bear with me here...

Do you think this Allen Bradley Ethernet/IP adapter would be sufficient?  https://www.plchardware.com/Products/RA-1761-NET-ENI.aspx

Currently I'm assigning a static IP to the MicroLogix 1400 and then using that with pycomm.  I feel like Allen Bradley would not sell these adapters if they did not communicate with the PLC the same way a built in ethernet port does.

I will hopefully be getting one of these soon and I can update you on how that goes if you like

magiciin, 2020-08-20:
Any update on using 1761-NET-ENI and communication with PYCOMM?