ruscito/pycomm #63: Reading Hundreds of Tags Fast

restyl1997, 2020-02-16:
First, great library, thank you for maintaining.

I'm trying to read ~700 of tags at once:
`import time
from pylogix import PLC

read = True
while read:
	try:
		with PLC() as comm:
			comm.IPAddress = '192.168.1.10'
			start = time.time()
			tag1 = comm.Read('tag1')
			tag2 = comm.Read('tag2')
			tag3 = comm.Read('tag3')
			...
			end = time.time()
			print("Successfully polled all tags:", end-start)
			time.sleep(1)
	except KeyboardInterrupt:
		print('Exiting, CTRL + C was pressed on the keyboard.')
		read = False
		quit()`

The issue is that it takes 14 to 15 seconds to return all of the tag values, and I need the newest values for each tag every second. How could I speed this up?

restyl1997, 2020-02-16:
I solved this using map(), I think you should include this code as an example for other people who run into this issue:

# Read value of tag in PLC
def readVal(tag): 
	return comm.Read(tag) 	
	
read = True
while read:
	try:
		with PLC() as comm:
			comm.IPAddress = '192.168.1.10'
			map(readVal, tags)
	except KeyboardInterrupt:
		print('Exiting, CTRL + C was pressed on the keyboard.')
		read = False
		quit()

ruscito, 2020-02-17:
Hello thanks for the suggestion I will add to the example page :)

closed 2020-02-17 by ruscito